adafruit/Adafruit_Python_SSD1306 #9: Debian 2015-03-01?

RGB-123, 2015-07-20:
Mike, I am trying to run this code on a fresh install of Debian and get the following errors when running the examples.  Any chance you have tried code on a BBB recently?

```
root@beaglebone:~/Adafruit_Python_SSD1306/examples# python animate.py
Traceback (most recent call last):
  File "animate.py", line 74, in <module>
    font = ImageFont.load_default()
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFont.py", line 377, in l                                           oad_default
    '''))))
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFont.py", line 117, in _                                           load_pilfont_data
    image.load()
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFile.py", line 189, in l                                           oad
    d = Image._getdecoder(self.mode, d, a, self.decoderconfig)
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 385, in _getd                                           ecoder
    raise IOError("decoder %s not available" % decoder_name)
IOError: decoder zip not available

tdicola, 2015-07-20:
Hrm, for the zip decoder not found error it looks like PIL might not have found the zip decoding library when it was installed.  Try the steps mentioned in this [stackoverflow answer](http://stackoverflow.com/a/16908700) to uninstall it, add a few zip, jpeg, etc. libraries, and then reinstall so it hopefully picks up those libraries:
```
pip uninstall PIL
apt-get install libjpeg-dev
apt-get install libfreetype6-dev
apt-get install zlib1g-dev
apt-get install libpng12-dev
pip install PIL --upgrade
```
(if you don't have pip installed then `sudo apt-get install python-pip` first should grab it)  Let me know if that fixes the issue, thanks!

RGB-123, 2015-07-22:
Mmmm, no luck there.  Most of the packages were up to date. Python-pip was also already installed and up to date.

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada