adafruit/Adafruit_Python_SSD1306 #12: banana pi TypeError: argument 1 must be string, not int (SSD1306)

norrest, 2016-07-12:
python -V
Python 2.7.9

uname -a
Linux 4.4.8 #3 SMP Sun Apr 24 14:46:20 CEST 2016 armv7l GNU/Linux

sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

i2cdetect -l
i2c-0   i2c             mv64xxx_i2c adapter                     I2C adapter
i2c-1   i2c             mv64xxx_i2c adapter                     I2C adapter

gpio readall
+----------+-Rev3-+------+--------+------+-------+
| wiringPi | GPIO | Phys | Name   | Mode | Value |
+----------+------+------+--------+------+-------+
|      0   |  17  |  11  | GPIO 0 | IN   | Low   |
|      1   |  18  |  12  | GPIO 1 | IN   | Low   |
|      2   |  27  |  13  | GPIO 2 | IN   | Low   |
|      3   |  22  |  15  | GPIO 3 | IN   | Low   |
|      4   |  23  |  16  | GPIO 4 | IN   | Low   |
|      5   |  24  |  18  | GPIO 5 | IN   | Low   |
|      6   |  25  |  22  | GPIO 6 | IN   | Low   |
|      7   |   4  |   7  | GPIO 7 | IN   | Low   |
|      8   |   2  |   3  | SDA    | ALT5 | Low   |
|      9   |   3  |   5  | SCL    | ALT5 | Low   |
|     10   |   8  |  24  | CE0    | ALT5 | Low   |
|     11   |   7  |  26  | CE1    | ALT5 | Low   |
|     12   |  10  |  19  | MOSI   | ALT5 | Low   |
|     13   |   9  |  21  | MISO   | ALT5 | Low   |
|     14   |  11  |  23  | SCLK   | ALT5 | Low   |
|     15   |  14  |   8  | TxD    | ALT0 | Low   |
|     16   |  15  |  10  | RxD    | ALT0 | Low   |
|     17   |  28  |   3  | GPIO 8 | IN   | Low   |
|     18   |  29  |   4  | GPIO 9 | ALT4 | Low   |
|     19   |  30  |   5  | GPIO10 | OUT  | High  |
|     20   |  31  |   6  | GPIO11 | ALT4 | Low   |
+----------+------+------+--------+------+-------+

sudo python shapes.py
Traceback (most recent call last):
  File "shapes.py", line 52, in <module>
    disp = Adafruit_SSD1306.SSD1306_128_64(rst=RST, i2c_address=0x3C)
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 240, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 88, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
TypeError: argument 1 must be string, not int

/lcd/examples# sudo python shapes.py
Traceback (most recent call last):
  File "shapes.py", line 52, in <module>
    disp = Adafruit_SSD1306.SSD1306_128_64(rst=RST, i2c_address=0x3C)
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 240, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 88, in __init__
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/GPIO.py", line 278, in setup
TypeError: argument 1 must be string, not int

closed 2016-07-12 by norrest

luxin88, 2017-03-15:
how did you solve that?

hadifallahi, 2017-05-23:
TypeError: argument 1 must be string, not int    i have this problem in my adafruit lcd examles but don,t know how to slove it

zzz3000, 2018-08-28:
Only example which works for me is stats.py. It has RST = None
Other don't work. They has RST = 24. 
RST = 24 cause "TypeError: argument 1 must be string, not int"