adafruit/Adafruit_Python_SSD1306 #15: TypeError: __init__() takes at least 3 arguments (3 given)

wojtulab, 2017-04-26:
Hello,
I have following issue on RBPI:

```
pi@RBPI:~/apps/OLED/python-examples $ sudo python OLEDclock.py
Traceback (most recent call last):
  File "OLEDclock.py", line 18, in <module>
    led = gaugette.ssd1306.SSD1306(reset_pin=RESET_PIN, dc_pin=DC_PIN)
TypeError: __init__() takes at least 3 arguments (3 given)
```

OS:
Linux RBPI 4.4.21-v7+ #911 SMP Thu Sep 15 14:22:38 BST 2016 armv7l GNU/Linux with Python 2.7.9

wojtulab, 2017-05-07:
any feedback, help or hint?

Vilhjalmr26, 2017-05-28:
you should be creating an item of the typ eof display you have, so ssd1306.SSD1306_128_64 or ssd1306.SSD1306_128_32.  That may help.

wojtulab, 2017-06-03:
ok i see. I will try it and back to U.

wojtulab, 2017-06-07:
hello, another issue:
[image]

Vilhjalmr26, 2017-06-07:
Sorry, missed this earlier. For using this library I would expect your import to be
import Adafruit_SSD1306
Not
import gaugette.ssd1306
Unless you have a different setup

wojtulab, 2017-06-08:
changed to
[image]

another issue:
[image]

Vilhjalmr26, 2017-06-12:
Sorry, not seeing what the issue is here

wojtulab, 2017-06-12:
what do you mean?

this is my error:
[image]

my full config is:
[image]

Vilhjalmr26, 2017-06-12:
Sorry, my email hid the second image in the original request, hence my
confusion.
reset_pin should instead be rst, dc_pin should instead be dc
I suggest looking in the examples folder at the shapes.py file, it sets
everything up for a basic display, and you can modify from there.

wojtulab, 2017-06-13:
ok, changed to:
led = Adafruit_SSD1306.SSD1306_128_64(rst=RESET_PIN, dc=DC_PIN)

and another issue:
pi@RBPI:~/apps/Adafruit_Python_SSD1306/examples $ sudo python OLEDclock.py
Traceback (most recent call last):
  File "OLEDclock.py", line 14, in <module>
    led.begin()
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 147, in begin
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 244, in _initialize
  File "build/bdist.linux-armv7l/egg/Adafruit_SSD1306/SSD1306.py", line 128, in command
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO-1.0.0-py2.7.egg/Adafruit_GPIO/I2C.py", line 114, in write8
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_PureIO-0.2.0-py2.7.egg/Adafruit_PureIO/smbus.py", line 236, in write_byte_data
IOError: [Errno 5] Input/output error

[image]

ladyada, 2017-06-13:
hiya we're going to close this issue because it seems to be a tech support thread not software related. thx :)

closed 2017-06-13 by ladyada

wojtulab, 2017-06-13:
ok i see, sorry for that

sydneykcho, 2019-02-12:
I faced the same issue as above case when running with Beaglebone Black.
If it be classified as a tech support thread, but it occurred on different platform and devices?