adafruit/Adafruit_Python_SSD1306 #16: RuntimeError: Could not determine platform

frankenstein91, 2017-06-04:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Pi zero Rev1 with Raspian and newst update) 

- Python version (run `python -version` or `python3 -version`):  .2.7.9 and 3.4.2

- Error message you are receiving, including any Python exception traces:  Traceback (most recent call last):
  File "Feldbahn/Fernbedienung.py", line 26, in <module>
    disp = Adafruit_SSD1306.SSD1306_128_64(rst=4, i2c_address=0x3d)
  File "build/bdist.linux-armv6l/egg/Adafruit_SSD1306/SSD1306.py", line 240, in __init__
  File "build/bdist.linux-armv6l/egg/Adafruit_SSD1306/SSD1306.py", line 85, in __init__
  File "build/bdist.linux-armv6l/egg/Adafruit_GPIO/GPIO.py", line 426, in get_platform_gpio
RuntimeError: Could not determine platform.

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
run one of the examples

vytasek, 2017-07-12:
Hi @frankenstein91, try to update Adafruit-GPIO to version 1.0.3. Just had a same issue today.

sydneykcho, 2019-02-12:
I faced the same issues when run the programs with Beaglebone Black.
And hope get help to solve the issues.  
Regards.

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada