adafruit/Adafruit_Python_SSD1306 #17: pip library not updated to latest

zaforic, 2017-11-15:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  
Raspbian: Linux ##### 4.9.61-v7+ #1049 SMP Fri Nov 10 15:32:51 GMT 2017 armv7l GNU/Linux
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch

- Python version (run `python -version` or `python3 -version`):  
pip: 9.0.1
python: 2.7.13

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "stats.py", line 29, in <module>
    disp = Adafruit_SSD1306.SSD1306_128_32(rst=RST)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_SSD1306/SSD1306.py", line 285, in __init__
    gpio, spi, i2c_bus, i2c_address, i2c)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_SSD1306/SSD1306.py", line 88, in __init__
    self._gpio.setup(self._rst, GPIO.OUT)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/GPIO.py", line 188, in setup
    pull_up_down=self._pud_mapping[pull_up_down])
ValueError: Channel must be an integer or list/tuple of integers

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

pip install Adafruit-GPIO Adafruit-SSD1306
apt-get install python-pilkit
apt-get install python-imaging

Run the script below and you will get the error message. I replaced the SSD1306.py script with the one from this repo and everything works. It appears the pip repo is out of date from this repo.

****************************************** CODE *****************************************
import time
import Adafruit_GPIO.SPI as SPI
import Adafruit_SSD1306
import subprocess
import signal
import sys

from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont

def signal_term_handler(signal, frame):
    disp.command(Adafruit_SSD1306.SSD1306_DISPLAYOFF)
    sys.exit(0)

#Setup Signal Handler
signal.signal(signal.SIGTERM, signal_term_handler)

# Raspberry Pi pin configuration:
RST = None     # on the PiOLED this pin isnt used

# Note the following are only used with SPI:
DC = 23
SPI_PORT = 0
SPI_DEVICE = 0

# 128x32 display with hardware I2C:
disp = Adafruit_SSD1306.SSD1306_128_32(rst=RST)

# Initialize library.
disp.begin()

# Clear display.
disp.clear()
disp.display()

# Create blank image for drawing.
width = disp.width
height = disp.height
image = Image.new('1', (width, height))

# Get drawing object to draw on image.
draw = ImageDraw.Draw(image)

# Draw a black filled box to clear the image.
draw.rectangle((0,0,width,height), outline=0, fill=0)

# Draw some shapes.
padding = -2
top = padding
bottom = height-padding
x = 0

# Load default font.
font = ImageFont.load_default()

try:
    while True:

        # Draw a black filled box to clear the image.
        draw.rectangle((0,0,width,height), outline=0, fill=0)

        # Shell scripts for system monitoring from here : https://unix.stackexchange.com/questions/119126/command-to-display-memory-usage-disk-usage-and-cpu-load
        cmd = "hostname -I | cut -d\' \' -f1"
        IP = subprocess.check_output(cmd, shell = True )

        cmd = "top -bn1 | grep load | awk '{printf \"Load:%.2f\", $(NF-2)}'"
        CPU = subprocess.check_output(cmd, shell = True )

        cmd = " echo $(($(cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq)/1000))"
        Freq = subprocess.check_output(cmd, shell = True )

        cmd = "free -m | awk 'NR==2{printf \"Mem: %s/%sMB %.2f%%\", $3,$2,$3*100/$2 }'"
        MemUsage = subprocess.check_output(cmd, shell = True )

        cmd = "cat /sys/class/thermal/thermal_zone0/temp"
        CPUTemp = subprocess.check_output(cmd, shell = True )
        CPUTemp = int(CPUTemp)/1000
        CPUTemp = str(9.0/5.0 * CPUTemp + 32) + "F"

        cmd = "vcgencmd measure_temp | sed 's/temp=//g'"
        GPUTemp = subprocess.check_output(cmd, shell = True )
        GPUTemp = str(GPUTemp).replace("temp=","").replace("'C","")
        GPUTemp = str(9.0/5.0 * float(GPUTemp) + 32) + "F"

        draw.text((x, top),       "IP: " + str(IP),  font=font, fill=255)
        draw.text((x, top+8),     str(CPU) + "  Freq:" + str(Freq), font=font, fill=255)
        draw.text((x, top+16),    str(MemUsage),  font=font, fill=255)
        draw.text((x, top+25),    "CPU:" + str(CPUTemp) + " GPU:" + str(GPUTemp),  font=font, fill=255)

        # Display image
        disp.image(image)
        disp.display()
        time.sleep(.5)
except KeyboardInterrupt:
    disp.command(Adafruit_SSD1306.SSD1306_DISPLAYOFF)

steverweber, 2018-02-22:
also hit this issue... 
thanks for reporting it ow i would still be stuck.
after replacing /usr/local/lib/python3.5/dist-packages/Adafruit_SSD1306/SSD1306.py with the git version i was up and running.

Please update your pip package so your users have a clean experience.

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada