adafruit/Adafruit_Python_SSD1306 #20: multiple oled display to single raspberry pi

rishikeshayre1310, 2018-01-19:
Hello
is it possible to interface multiple oled display through i2c. and if so what changes need to be done.

MarkFiechtner, 2018-04-28:
I don't see a way to set the I2C address on the SSD1306 so you might need a unique I2C bus per device.  You didn't say what type of processor board you are using so it would depend on your hardware.
The 1306 data sheet is here if you want to look into it:
https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf
Good luck.

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada