adafruit/Adafruit_Python_SSD1306 #22: "File is not a zip file" error when building per website instructions

PaulSlocum, 2018-04-19:
Using instructions from website here: https://learn.adafruit.com/adafruit-pioled-128x32-mini-oled-for-raspberry-pi/usage

On latest Raspbian Stretch:

pi@raspberrypi:~ $ git clone https://github.com/adafruit/Adafruit_Python_SSD1306.git
Cloning into 'Adafruit_Python_SSD1306'...
remote: Counting objects: 112, done.
remote: Total 112 (delta 0), reused 0 (delta 0), pack-reused 112
Receiving objects: 100% (112/112), 34.60 KiB | 0 bytes/s, done.
Resolving deltas: 100% (57/57), done.
pi@raspberrypi:~ $ cd Adafruit_Python_SSD1306
pi@raspberrypi:~/Adafruit_Python_SSD1306 $ sudo python setup.py install
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-3.5.1.zip
Extracting in /tmp/tmpPeTB4Y
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/Adafruit_Python_SSD1306/ez_setup.py", line 128, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/Adafruit_Python_SSD1306/ez_setup.py", line 108, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/Adafruit_Python_SSD1306/ez_setup.py", line 57, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/Adafruit_Python_SSD1306/ez_setup.py", line 88, in archive_context
    with get_zip_class()(filename) as archive:
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file

brennen, 2018-05-15:
After trying in a couple of environments (latest Raspbian image, and stock Debian Stretch), I'm unable to reproduce this behavior.

I do however notice a couple of redirects here:

```
pi@raspberrypi:~/Adafruit_Python_SSD1306 $ wget https://pypi.python.org/packages/source/s/setuptools/setuptools-3.5.1.zip
--2018-05-15 19:53:53--  https://pypi.python.org/packages/source/s/setuptools/setuptools-3.5.1.zip
Resolving pypi.python.org (pypi.python.org)... 151.101.0.223, 151.101.64.223, 151.101.128.223, ...
Connecting to pypi.python.org (pypi.python.org)|151.101.0.223|:443... connected.
HTTP request sent, awaiting response... 301 Redirect to Primary Domain
Location: https://pypi.org/packages/source/s/setuptools/setuptools-3.5.1.zip [following]
--2018-05-15 19:53:54--  https://pypi.org/packages/source/s/setuptools/setuptools-3.5.1.zip
Resolving pypi.org (pypi.org)... 151.101.64.223, 151.101.0.223, 151.101.192.223, ...
Connecting to pypi.org (pypi.org)|151.101.64.223|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://files.pythonhosted.org/packages/source/s/setuptools/setuptools-3.5.1.zip [following]
--2018-05-15 19:53:54--  https://files.pythonhosted.org/packages/source/s/setuptools/setuptools-3.5.1.zip
Resolving files.pythonhosted.org (files.pythonhosted.org)... 151.101.1.63, 151.101.65.63, 151.101.129.63, ...
Connecting to files.pythonhosted.org (files.pythonhosted.org)|151.101.1.63|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://files.pythonhosted.org/packages/6f/6f/c26e40e5ffa9aa4601d9fa27a7238ef38bf15d19e683a5edb2524cf156ab/setuptools-3.5.1.zip [following]
--2018-05-15 19:53:55--  https://files.pythonhosted.org/packages/6f/6f/c26e40e5ffa9aa4601d9fa27a7238ef38bf15d19e683a5edb2524cf156ab/setuptools-3.5.1.zip
Reusing existing connection to files.pythonhosted.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 926881 (905K) [application/octet-stream]
Saving to: ‘setuptools-3.5.1.zip’

setuptools-3.5.1.zip                  100%[========================================================================>] 905.16K  2.08MB/s    in 0.4s    

2018-05-15 19:53:55 (2.08 MB/s) - ‘setuptools-3.5.1.zip’ saved [926881/926881]
```

...so I can imagine something failing to follow those correctly.  Lemme see if I can adjust to the correct `setuptools-3.5.1.zip` URL.

DuncDude, 2018-05-15:
Im also having this problem...

brennen, 2018-05-16:
This seems, from googling, like a pretty common class of bugs.  I just pushed a branch updating `ez_setup.py` to a newer release; can anyone who's seeing the issue try reproducing with that branch?

```sh
git clone https://github.com/adafruit/Adafruit_Python_SSD1306.git
cd Adafruit_Python_SSD1306
git checkout update_ez_install
sudo python setup.py install
```

brennen, 2018-05-17:
I just pushed another change here.  Per commentary on some other repos and the setuptools docs, let's try this without ez_setup.py involved at all:

```
sudo python -m pip install --upgrade pip setuptools wheel
git clone https://github.com/adafruit/Adafruit_Python_SSD1306.git
cd Adafruit_Python_SSD1306
git checkout update_ez_install
sudo python setup.py install
```

brennen, 2018-05-17:
Still curious if that works, but looking at it, I think we're gonna move our install instructions generally to just use `pip` for the whole deal.

DuncDude, 2018-05-17:
Alright I burned a fresh copy of Jessie just in case, updated, updated,python, installed pip, installed github, cloned the repository, did the git checkout. Now errors are as follows:
`pi@raspberrypi:~/Adafruit_Python_SSD1306 $ sudo python -m pip install --upgrade pip setuptools wheel
/usr/bin/python: No module named pip

`
and 
`pi@raspberrypi:~/Adafruit_Python_SSD1306 $ sudo python setup.py install
Traceback (most recent call last):
  File "setup.py", line 1, in <module>
    from setuptools import setup, find_packages
ImportError: No module named setuptools
`

maybe its directed to the wrong location for pip?

brennen, 2018-05-21:
@DuncDude maybe. Does `sudo pip install setuptools` get you anywhere?  I also just noticed that you're running Jessie rather than Stretch, which is likely to be a complicating factor...

danjal, 2018-05-24:
I get the exact same error as described in the beginning of the issue.
I use Raspbian Stretch Lite (2018-04-18).

It worked on Raspbian Stretch Lite (2017-09-07).

GIJ0E, 2018-07-18:
I was trying to install the Adafruit PCA9685 and ran across the same problem in this thread about the bad zip file.
Here's what I did to get things going...
When I looked at the setuptools-3.5.1.zip that it downloaded automatically, the zip file had the '301 moved permanently' HTML as mentioned above by braydenm303. Apparently it didn't redirect as brennon's did above. 
So, to get the real zip file, I deleted the bad zip file and manually downloaded the zip file using 
`wget https://pypi.python.org/packages/source/s/setuptools/setuptools-3.5.1.zip`
After it was complete, I re-ran the python install command and it worked.

This was in Raspbian Stretch Lite (2017-11-29).
Hope this helps!!

bill2009, 2018-09-30:
@GIJ0E : This worked for me.  recommend simplifying the install and move this to the top of the thread.

boeremak, 2018-11-07:
@GIJ0E That was it, thanks man. Seems like the installer script does not know how to deal with redirects.

meeki007, 2018-11-14:
@GIJ0E : Thanks for the fix. I also had to use your method

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada