adafruit/Adafruit_Python_SSD1306 #31: IOError: [Errno 2] No such file or directory: '/dev/i2c-1'

NirajSudhirMehta, 2019-03-12:
Hardware : Orange Pi Zero

OS : ARMBIAN 5.73 stable Debian GNU/Linux 9 (stretch) 4.19.20-sunxi

Python : Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2

sudo i2cdetect -y 0
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

python
Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.

            import Adafruit_GPIO.SPI as SPI
            import Adafruit_SSD1306
            RST = None
            DC = 23
            SPI_PORT = 0
            SPI_DEVICE = 0
            disp = Adafruit_SSD1306.SSD1306_128_64(rst=RST, i2c_address=0x3C)
            Traceback (most recent call last):
            File "", line 1, in
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_SSD1306/SSD1306.py", line 243, in init
            gpio, spi, i2c_bus, i2c_address, i2c)
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_SSD1306/SSD1306.py", line 107, in init
            self._i2c = I2C.get_i2c_device(i2c_address)
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/I2C.py", line 64, in get_i2c_device
            return Device(address, busnum, i2c_interface, **kwargs)
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_GPIO/I2C.py", line 97, in init
            self._bus = Adafruit_PureIO.smbus.SMBus(busnum)
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_PureIO/smbus.py", line 105, in init
            self.open(bus)
            File "/usr/local/lib/python2.7/dist-packages/Adafruit_PureIO/smbus.py", line 130, in open
            self._device = open('/dev/i2c-{0}'.format(bus), 'r+b', buffering=0)
            IOError: [Errno 2] No such file or directory: '/dev/i2c-1'

ZhuNathan, 2019-08-30:
Enable the i2c in system configuration

ladyada, 2019-09-27:
hiya, this library is being deprecated - we have a similar library that works with Python3 and modern linux computers. please check out this guide for more info
https://learn.adafruit.com/monochrome-oled-breakouts/python-wiring

closed 2019-09-27 by ladyada